clariusdev/imu #1: Motion Problem

MarcoTripoli, 2021-07-01:
Hi,

I am working on a project using your motion project. I followed the various steps marked in the readme to start the project, but once started and connected to the probe, the 3d model does not appear on the screen, remaining on a white screen as shown below

[image]

the linux shell gives the following error

[image]

What could be the problem ?

Thanks in advance

clariusk, 2021-07-04:
you can learn more about Qt plugins here: https://doc.qt.io/qt-5/linux-deployment.html#qt-plugins

MarcoTripoli, 2021-07-05:
Hello,
I found that the problem stems from an unresolved Qt bug, as mentioned in the following link.

https://stackoverflow.com/questions/55637096/osx-with-qt3d-5-12-2-and-qml-sceneloader-found-no-suitable-importer-plugin-for

Unfortunately, the "solution" that is given does not work.

How did you get the Motion project to work on linux?

Thanks in advance

clariusk, 2021-07-05:
you're correct, I was testing on Windows and it was okay, but Linux, this bug seems to persist. you could write in some native OpenGL I'm guessing, but not sure it's worth your while if you're just looking to test out the IMU functionality as a display.

closed 2021-07-07 by clariusk