clariusdev/imu #2: Innacurate IMU captor

eclarius, 2022-03-30:
Hello,

We are testing the IMU sensor of the probe. After calibrating the probe with the app, we did a rotation of exactly 90 degree and back twice on our workbench and recorded the rotational speed. We simply integrated this speed to obtain the orientation:

[image]

Unfortunately we obtain 75 degrees of rotation, which is a >15% error. Such error on an easy and short task makes the IMU data worthless on longer capture. To verify that this is not due to our code, we attached an iPhone on our workbench to capture simultaneously the same data (up to axis and directions):

[image]

The integrated rotation gives us indeed 90 degrees:
[image]

- Is there anyone using IMU data from the probe successfully?
- Is there a better way to calibrate the sensor?
- Is our probe defect?

Thank you for your help

eclarius, 2022-03-31:
We are using the raw gyroscope data.

clariusk, 2022-03-31:
ok great, the calibration will not have an effect then. once calibrated you can then use the quaternion data (if that helps with your positioning calculations) - you can see an example of how that data is used: https://github.com/clariusdev/motion/blob/master/pyimu/pyimu.py

eclarius, 2022-04-01:
Great! That's much better :)

To understand better: the quaternions are computed from the gyroscope, accelerometer and magnetometer on the chip after calibration? Does that mean we should discard gyroscope and magnetometer data? Is the accelerometer calibrated and does it make sense to use it to find the (relative) position by double integration?

[image]

clariusk, 2022-04-01:
- yes, the quaternion data is calculated from the raw data with calibration data applied.
- not sure if you need to discard everything, for our basic rotation algorithms, we just use the calibrated data, but if you're looking to do translation, perhaps there's more methods that need to be applied on the raw data (sorry the spatial transformations are not my background)

closed 2022-04-22 by clariusk