clariusdev/imu #3: IMU captor position

eclarius, 2022-03-30:
Please provide the distance from scanning plane to internal sensor for the new scanner model (C3HD). This is needed to apply rotation quaternion to the CAD model. Finding this position ourselves is too difficult given inaccuracies in #2 .

clariusk, 2022-03-30:
is this for HD or HD3 model, if HD, then the internal CAD is provided here: https://github.com/clariusdev/cad/blob/main/hd.zip

eclarius, 2022-03-31:
Thanks for your answer.

We have the HD. I checked the CAD file and it does not contain any electronical component. There is also an `imu.step` file but I don't think it's for the C3HD.

[image]

Let us know where is the IMU sensor in our probe.

clariusk, 2022-03-31:
you can actually use the imu.step file, ignore the model of the scanhead, all of our devices have the same inside

eclarius, 2022-04-01:
Is it at the center of the red rectangle?

[image]

It would be much more reliable for every users if you indicated the position of the captor for all the HD and HD3 probes as you did for your v1.

[image]

clariusk, 2022-04-04:
check out the opposite side, the small solid rectangle.
[image]

eclarius, 2022-04-05:
Thanks for your reply. According to our measurement the captor is at -116mm from the imaging plane on the z-axis and we will assume it is centered on the x- and y-axis.

closed 2022-04-05 by eclarius