clariusdev/imu #6: Timestamps for imu.yml

benjaminralex, 2022-08-14:
I have the following questions for the attached yml:

1) Are the timestamps correct? When decoding it from posix to date/time format or when looking at the difference between time stamps, it seems to be erroneous.

2) Is there a nominal sensor frequency?

3) Does clarius provide an API or any code to extraction probe position in addition to the orientation?

Thanks
[imu_yml_as_text.txt](https://github.com/clariusdev/motion/files/9332543/imu_yml_as_text.txt)
!

clariusk, 2022-08-14:
1. yes look good, the stamping is done in an asynchronous process, so there will be some variance, but in general the data is read by the chip @ 100-200 Hz, and any data read within the software is stamped there - if you correlate with
 
2. 100-200Hz for reading of data as mentioned above, for other specs you can check out the data sheet: https://www.st.com/resource/en/datasheet/lsm9ds1.pdf

3. we have not worked on positioning with the imu, just orientation based on the quaternion data

benjaminralex, 2022-08-14:
Thank you so much for the information! 

But how would one go about acquiring the orientation for a given image if the timestamps are asynchronous? Do the images have asynchronous time stamps as well? If so, where can they be found?

julien-l, 2022-08-15:
The timestamps do not correspond to a particular time point (like Epoch) but all the timestamps for all the data acquired during a particular session all have the same (arbitrary) reference therefore you can correlate the orientation data to a particular image (you might have to interpolate if there is no data for a particular timestamp).

benjaminralex, 2022-08-15:
Makes sense. I can readily find the pose timestamps but can't seem to locate the image timestamps.. which file would contain those?

clariusk, 2022-08-15:
The timestamps are located in the first 64 bits of each image in a raw data file, see more at: https://github.com/clariusdev/raw#raw-data-formats

benjaminralex, 2022-08-16:
Thank you! Is there a way to estimate the sensor sampling rate from the asynchronous time stamps?

closed 2022-08-17 by benjaminralex