clariusdev/imu #10: IMU Data: Roll Pitch Yaw

km414, 2023-03-02:
Dear Clarius Support Engineers,

I have been trying to make sense of the imu.yml file for quite some time now, specifically regarding the roll pitch yaw measurements. 

2023Feb28RollPitchYawTest describes tests I have carried out to determine the accuracy of the roll pitch yaw measurements.
MATLAB code, ClariusRollPitchYaw.m, was used to analyze the imu data.
The [box link](https://app.box.com/s/z8j6k5zl1mcutfd6mdov6m4n33rckl7s) will allow you to access Test1-Test6 data files. 

Stationary calibration was performed as described: https://github.com/clariusdev/motion. 
I've referred frequently to the GitHub page repositories: https://github.com/orgs/clariusdev/repositories. 

The imu and raw data collection stops after the first ultrasound image is captured. Therefore, no ultrasound images were captured during these tests in the calibrated position. Only a single image was captured at the end of each test while the probe was in its final position.

The imu and raw data collection stops after the first ultrasound image is captured. (Therefore, no ultrasound images were captured during these tests in the calibrated position. Only a single image was captured at the end of each test while the probe was in its final position.) The results of the tests are inconsistent and I have not been able to extract any substantial information from the imu.yml file. 
Please answer the following questions: 

1. Is it normal that the imu data collection stops after capturing the first ultrasound image? Should this be happening?
2. Why are the initial positions/angles for the roll, pitch, and yaw in the imu data changing each trial if I am completing a stationary calibration prior to scanning? I do not touch move the scanner from its calibrated position for the first few seconds of each test. Shouldn’t these values start at or close to zero in this case? 
3. Is there something I am missing with the calibration procedure? Do I need to carry out a motion calibration? If so, could you please provide a more detailed description of the motion calibration process? 
4. Why do very sharp peaks appear in my time versus angle plots occasionally? How can I prevent this? 
5. Why are we getting such poor, inconsistent data? Is it something that I am doing incorrectly on my end? Is there a bug on Clarius’s end? 
6. Will the quality or accuracy of images or data decrease if I use Clarius Cast?  
7. Could the offset physical location of the IMU impact my results?   (The imu sensor in the PA HD is offset by (x, y, z) = (-115.70, 2, 6.11) mm.) 

The Clarius PA HD ultrasound probe has the research package, including raw data and 3D positional data:

We see other groups have gotten meaningful results from the IMU, so please advise. 
Thank you for the help.

Attachment(s)
[2023Feb28RollPitchYawTest.pdf](https://support.clarius.com/attachments/token/mfEiooLova4KE9YMwCjfe3cf5/?name=2023Feb28RollPitchYawTest.pdf)
[2023Feb28RollPitchYawTest.docx](https://support.clarius.com/attachments/token/Fyfg5AEZ44f4E0LAjxqTOOlqs/?name=2023Feb28RollPitchYawTest.docx)
[ClariusRollPitchYaw.m](https://support.clarius.com/attachments/token/vErs6Kpg0CuzHmxDmX60vHSox/?name=ClariusRollPitchYaw.m)

julien-l, 2023-03-02:
1. Yes
2. If the calibration changes between scans, then the IMU output will be different for each scan. Try to calibrate only once before your tests.
3. Perform both stationary and motion calibration as they both impact the IMU output. For motion calibration, take the scanner in your hand and move so that it rotates around all 3 axes.
4. Not sure what are the sharp peaks, you could consider them as outliers and remove them.
5. Try performing both stationary and motion calibration once before your scans. No reported bug with IMU acquisition.
6. Clarius Cast uses the same functions as the Clarius App -> no accuracy change.
7. The sensor location is used to correlate the images with the IMU data.

clariusk, 2023-03-02:
Hi there, sorry for the delays, our scientist who was running our IMU program has left to Johns Hopkins for a PhD, so I don't know if I will be able to answer the questions with the best knowledge, but I'll try.

> 1. Is it normal that the imu data collection stops after capturing the first ultrasound image? Should this be happening?
No, with the IMU turned on, i will capture multiple data points per frame, when save a raw cine which will contain multiple frames, you will get a corresponding IMU data file with correlating timestamps to the images (or raw data). With the Cast API, you will get a stream of IMU 

> 2. Why are the initial positions/angles for the roll, pitch, and yaw in the imu data changing each trial if I am completing a stationary calibration prior to scanning? I do not touch move the [...]
You shouldn't have to calibrate each time, just once every few months is fine. The sensors are under a high degree of noise, therefore you will never get the exact same results each time.

> 3. Is there something I am missing with the calibration procedure? Do I need to carry out a motion calibration? If so, could you please provide a more detailed description of the motion calibration process?
As mentioned above, calibrating once is fine, if it passes, the values have been written into the device.

> 4. Why do very sharp peaks appear in my time versus angle plots occasionally? How can I prevent this?
It may be due to the system noise, which you could try filtering out.

> 5. Why are we getting such poor, inconsistent data? Is it something that I am doing incorrectly on my end? Is there a bug on Clarius’s end?
It may be due to the system noise, which you could try filtering out.

> 6. Will the quality or accuracy of images or data decrease if I use Clarius Cast?
No, the Cast API streams the data, but does not have any affect on the collection or frame rates.

> 7. Could the offset physical location of the IMU impact my results?   (The imu sensor in the PA HD is offset by (x, y, z) = (-115.70, 2, 6.11) mm.)
The position may be useful depending on what you are trying to do with the data, sorry not really my domain of know how to process the data.

On another note, I often test the pyimu program we have online here, and it seems to show the orientation of the scanner relatively accurately, may be something you want to study as well.

closed 2023-03-31 by clariusk