clariusdev/imu #13: Sampling Rate of Clarius L7HD3

Souganttika, 2023-07-13:
Hi,

I would like to know what is the sampling frequency/sampling rate of the Clarius L7HD3 scanner and how I can set it to a desirable frequency. Thank you.

Also, is it possible to change the way the timestamps are generated in the RAW data?

clariusk, 2023-07-13:
what is the preset and depth ranges you are looking at getting the frequency/sampling rate information for, they may vary?

The timestamps are generated from the probe's internal monotonic clock, it cannot be changed.

Souganttika, 2023-07-14:
I am looking at SC Compound mode at a depth of 4cm. The video is recorded for a duration of 5 seconds and the raw IMU data is extracted. Between two recordings of 5 seconds with the same presets, the size of data extracted largely varies. How can that be recified?
[Stationary_Test_1307_2.zip](https://github.com/clariusdev/motion/files/12047883/Stationary_Test_1307_2.zip)

Attached are the raw data extracted with the same settings.