clariusdev/imu #17: IMU Calibration in v11.1.0

deepakraina99, 2024-02-12:
Hi, I could not locate the option to calibrate the probe in the Android app version v11.1.0. Please guide. And, I am also getting _len(imu)_ in _pycaster.py_ as 0.

julien-l, 2024-02-12:
Find a guide here: https://github.com/clariusdev/motion

deepakraina99, 2024-02-12:
Thanks, calibration is successful for both stationary and motion. But IMU data is still empty in new image callback function. Anything else I missed?

deepakraina99, 2024-02-12:
I checked the imaging modes in the app, but could not locate the IMU icon to activate it.  See below
[image]

julien-l, 2024-02-12:
This must be a licensing issue. If you have the research package, you should have access to the research menu, like in my screenshot, but I do not see it on yours. In that case, please contact Clarius support https://support.clarius.com/, thanks!

deepakraina99, 2024-02-18:
This has been resolved by https://support.clarius.com/. Thanks.