clariusdev/imu #19: IMU data saving issues

schoenlebn, 2024-07-02:
Is it possible to save the IMU data locally after data collection, without also saving the iq data. My lab has been saving the IMU data to an Ipad and has been able to reproducibly acquire the IMU data when the Buffer Raw Data is also activated, but is otherwise unable. The advertising states it is stored and tagged alongside the video and image data, is that feature only available in Clarius cloud??

schoenlebn, 2024-07-02:
For what it matters, we are also unable to find the IMU data in Clarius cloud as well. Is there something we are doing wrong? If so, what?

taharh, 2025-04-30:
> For what it matters, we are also unable to find the IMU data in Clarius cloud as well. Is there something we are doing wrong? If so, what?

We are encountering same issue here. @clariusk any help please?

clariusk, 2025-04-30:
I'm not sure about the cloud IMU data, I am not able to look into it directly myself. Best methods for collecting IMU currently will be to:
- Capture the raw data with IMU turned on, then the raw data packages both locally and exported to cloud will have the information
- Use the Cast API / Python wrapper to obtain the IMU data in real-time alongside the images

thanks